AusTINCANsProgrammingTeam/2022RobotCode #14: Expand MotorController SmartDashboard output

azvanderpas, 2022-01-15:
https://github.com/AusTINCANsProgrammingTeam/2022RobotCode/blob/d5579f37169a2eb71b6de95731a66fed51a0eca2/src/main/java/frc/robot/common/hardware/MotorController.java#L70

Currently updateSmartDashboard just logs and allows writing PID constants.  We should add more debug information to the smartdashboard. 

Also, looks like there is a Glass UI widget specifically for PID controllers: https://docs.wpilib.org/en/stable/docs/software/dashboards/glass/widgets.html#pid-controller-widget

caseyjbrotherton, 2022-02-11:
There is a shuffleboard widget for PID controllers as well.  
Our motorController doesn't to implement or extend PIDController, or MotorController.  Fixing that might help with the standard widgets that we are able to use

sarango2158, 2022-03-23:
Making this a post-season goal to help us set up for next year. We should use the PID widget to easily tune.