AusTINCANsProgrammingTeam/2022RobotCode #23: Add secondary climb camera

sarango2158, 2022-01-19:
Climber will have a camera pointing directly upwards to align the robot. Will need to stream the feed to a dashboard for the drivers

sarango2158, 2022-02-16:
We'll have the limelight and a drive camera on the robot. We may need to attach the drive camera to a servo and tilt it up physically for climb.

sarango2158, 2022-02-19:
Possibly will make do without this, lowering to a stretch goal

Majorferguson21, 2022-02-22:
Do we have our camera ready in the shop? I can hopefully get this set up quickly if we do

sarango2158, 2022-03-23:
We should re-evaluate this issue when we have a traversal climb @ModBoyex @loipiop224

Majorferguson21, 2022-03-23:
After actually being at a competition and climbing, I don't know how good a climb camera would be, and I'm also a little concerned about the act of changing cameras from code. (At the moment, I run a powershell command to start mpv when setting the driver station up.