AusTINCANsProgrammingTeam/2022RobotCode #38: Integer division likely cancels out gravity component of projectile calculation

azvanderpas, 2022-01-27:
https://github.com/AusTINCANsProgrammingTeam/2022RobotCode/blob/85c9d208f1c986122bf16686a73a6b00fc7e33ae/src/main/java/frc/robot/subsystems/ShooterSubsystem.java#L125

The 1 / 2 of this line likely results as 0 since integer division will truncate any decimal/remainder. Either need to use 0.5 or at least one of the numbers needs to be a floating point literal, e.g. 1.0 / 2. 

@loipiop224 @Majorferguson21

caseyjbrotherton, 2022-02-10:
@loipiop224 We had talked about this as well.  I believe you changed the 1/2, but don't see it in the code.

Seems like an easier one to make correct and merge, so we can close this issue.

sarango2158, 2022-02-16:
Still an issue in latest for Shooter-Main(jim's) and ShooterTesting

Majorferguson21, 2022-02-19:
Closing this, we removed projectile prediction from the shooter's functionality and have moved to work with the lookup exclusively.

closed 2022-02-19 by Majorferguson21