AusTINCANsProgrammingTeam/2022RobotCode #40: Indicate to operator when robot is ready to shoot

sarango2158, 2022-01-28:
When the robot has at least one ball, is positioned, and is spun up for a shot, there should be an indicator to the human operator that they can press shoot. 

Options:
1. An indicator on the driver dashboard software
2. CAD created a physical sign we can use that will be lit by LEDs. Can connect that up to a DIO and turn the LEDs ON when the robot is ready to shoot.

CenterIce1919, 2022-01-28:
Yes, the sign is edge lit, and could be multi-color LEDs. You would have to get the value from the driver dashboard to a USB controller to power the 12V LED strip. We can integrate the signs into the operator console, so that they could be more in the driver's field of view. We can do signs for more then when to shoot, like to indicate what color cargo we are holding and in what order.

caseyjbrotherton, 2022-02-05:
Pull request above puts a boolean to smart dashboard.  Do we want to split off an issue to trigger the visual reference from CAD through USB?

caseyjbrotherton, 2022-02-10:
Closing this, as there is another issue regarding physical indicator when ready to shoot.