AusTINCANsProgrammingTeam/2022RobotCode #56: Physical Indicator to indicate when robot is ready to shoot

caseyjbrotherton, 2022-02-05:
Open from #40 to capture the physical indicator, as there is a pull request to add an indicator to the dashboard.

CAD created a physical sign we can use that will be lit by LEDs. Can connect that up to USB and turn the LEDs ON when the robot is ready to shoot.

Yes, the sign is edge lit, and could be multi-color LEDs. You would have to get the value from the driver dashboard to a USB controller to power the 12V LED strip. We can integrate the signs into the operator console, so that they could be more in the driver's field of view. We can do signs for more then when to shoot, like to indicate what color cargo we are holding and in what order.

caseyjbrotherton, 2022-02-09:
Have we thought through what it would look like to turn on the lights for the targeting reticle from CAD?

I saw that there was a way to extend widgets on [Shuffleboard](https://github.com/nleach999/Vortx3735.Shuffleboard.2019.WidgetTutorial)
Feel like that could be difficult to do

If there is a NetworkTable entry, then either 
[Java](https://docs.wpilib.org/en/stable/docs/software/networktables/client-side-program.html) or
 [python](https://robotpy.readthedocs.io/projects/pynetworktables/en/stable/examples.html) can read that entry and do something.

caseyjbrotherton, 2022-02-18:
We have this:
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h
as well as this:
https://www.adafruit.com/product/2241 ( tutorial: https://learn.adafruit.com/adafruit-dotstar-leds )

As possible resources when we are ready

Majorferguson21, 2022-02-22:
I feel like this should be a stretch, as cool of an idea as this is, especially as the program fires automatically

sarango2158, 2022-03-23:
It seems to me like this issue can be closed. Our shooter will automatically release cargo when the flywheel is at speed, and the drive team has requested not to use automated alignment for shooting. There doesn't seem to be a need for an indicator. @Majorferguson21 what do you think?

Majorferguson21, 2022-03-23:
I'm gonna close this, yeah

closed 2022-03-23 by Majorferguson21