AusTINCANsProgrammingTeam/2022RobotCode #57: Constants Conflicts

ModBoyex, 2022-02-05:
In the Constants file, there were two fives conflicting with each other.

sarango2158, 2022-02-05:
All constants need to be updated to their real values on the robot. 

- driveRightFront = 7
- driveRightRear = 8
- driveLeftFront = 1
- driveLeftRear = 2
- intakeMotorOneID = 3
- CDSBeltID = 6
- intakeWheelOneID = 12
- intakeWheelTwoID = 11
- shooterCargoID = 5
- shooterID = 4

sarango2158, 2022-02-05:
Joystick constants also in conflict:
    public static final int BButton =           2; // Button for forward cds
    public static final int AButton =           2; //Button to align Limelight

    public static final int downbutton =        4; // Button for shooter mode
    public static final int YButton =           4; // Button for reverse cds

    public static final int rightBumperButton = 5; // Button for intake
    public static final int upbutton =          5; // Button for shooter mode

sarango2158, 2022-02-11:
Still needs joystick constants cleaned up

closed 2022-02-12 by sarango2158