AusTINCANsProgrammingTeam/2022RobotCode #62: Debug Tab Creation Needs To Be Moved

ModBoyex, 2022-02-05:
Robot Container, Line 48. 

The Debug Tab needs to be created here before Subsystems Call it.

sarango2158, 2022-02-05:
Assigning to Shooter team as they use this in their subsystem

caseyjbrotherton, 2022-02-09:
There are two ways to resolve this:

1. Keep the create tab in the Robot Container, and make sure it is called, and passed to each subsystem.
2. Call getTab in each subsystem that wants to use a tab/

For details on the second, the tabs are stored as network table entries.  A call to getTab with the same tab will return the same handle:  https://frcdocs.wpi.edu/en/latest/docs/software/dashboards/shuffleboard/layouts-with-code/using-tabs.html

So, in this example code, the following two lines are referencing the same tab:
https://github.com/caseyjbrotherton/ShuffleBoardExamples/blob/c9dbbcddecd20eee6edb9917451a61f5877aa9c1/src/main/java/frc/robot/subsystems/ExampleSubsystem.java#L38
https://github.com/caseyjbrotherton/ShuffleBoardExamples/blob/da6d66e1712c87fb37c3b7ff47615cf1369f3000/src/main/java/frc/robot/subsystems/DriveSubsystem.java#L33

Each of them will return the same handle, and allow writes to the same tab.

sarango2158, 2022-02-16:
In main currently a debugTab instance is created before anything else inside RobotContainer. That should resolve this error.