AusTINCANsProgrammingTeam/2022RobotCode #73: DriveBase-Test Path following issue

AndrewIverson, 2022-02-11:
https://github.com/AusTINCANsProgrammingTeam/2022RobotCode/blob/8820c4f9ff6bf3b07d84f0fbe7754ce2061a4877/src/main/java/frc/robot/subsystems/DriveBaseSubsystem.java#L84

This should be divided by gear ratio.

Calculate our circumference `2 * Math.PI * Constants.wheelRadius`
Convert to meters ` / Constants.inchesInMeter`
That gives us meters per wheel rotation

Our encoder.GetPosition() is the native units multiplied by the positionConversionFactor,  In this case the native units are rotations
So we have distance per one rotation of the wheel. But we need distance per one rotation of the motor. We can get that by dividing by the gear ratio constant.

tonyn10, 2022-02-12:
I fixed the conversion. 

Is the pid controller we use similar to the internal relative encoder (records the pid values of the motor not the wheel)? If so, in the acceptWheelSpeeds biconsumer function, we would have to convert the value in terms of rpm for the motor?

tonyn10, 2022-02-12:
I have the gear ratio constant as 10.75. 
If the gear ratio is 10.75 wheel rotations : 1 motor rotation for the kitbot, 
wouldn't multiplying by 1/10.75 or dividing by 10.75 give motor rotations per minute? Is the ratio I have reversed?

AndrewIverson, 2022-02-12:
Its 10.75 motor rotations : 1 rotation of the wheel.

tonyn10, 2022-02-12:
Gotcha, I understand now. Thank you.

closed 2022-02-12 by tonyn10