AusTINCANsProgrammingTeam/2022RobotCode #95: Create a "competition controls" layout for button bindings

sarango2158, 2022-02-19:


sarango2158, 2022-02-19:
[image]
[image]

azvanderpas, 2022-02-28:
I think the CDS/intake mappings are confusing. They should be mapped to trigger together, right @sarango2158?

@ModBoyex can you document here what your one controller test mapping is here?

closed 2022-03-05 by azvanderpas

Majorferguson21, 2022-03-09:
Due to driver requests, I have rebinded:
-Main shooter mode (currently TARMAC, LOW/AUTO will likely not be necessary/operable) to left trigger
-Secondary shooter mode (LOW at the moment) to left bumper
-Intake to the right trigger
-Outtake to the right bumper