AusTINCANsProgrammingTeam/2022RobotCode #125: I2C Lockup

azvanderpas, 2022-02-28:
Having I2C accesses every scheduler cycle even while our robot is disabled seems like a likely source of the lockups. 

If an I2C transaction is in the middle of running when we redeploy, the I2C clock (SCL) could stop in the middle of read transaction where the target device is driving low. This prevents the roboRIO from properly issuing a new start command. 

I2C SDA is bidirectional where a '0' means a device is actively driving it to GND and a '1' is disconnected with a pullup resistor biasing the line to VCC. So if the color sensor or mux device is driving '0' when the code is stopped the roboRIO may see the sensor as 'busy'. 

Prevention: Stop I2C devices before redeploying. Maybe close the I2C port when we disable the robot? Need to ensure it completes any pending transactions rather than just killing it. 

Mitigation: Force I2C interface to send 10 or more SCL pulses on code start. This is easier to do on the MXP port since it can be configured as a DIO pin and toggled directly.

azvanderpas, 2022-03-01:
https://github.com/AusTINCANsProgrammingTeam/2022RobotCode/blob/2c2c625026f3e7ad890ffa4118fffaf18b40666d/src/main/java/frc/robot/subsystems/CDSSubsystem.java#L191

The CDSSubsystem currently checks all the colors in it's periodic() method. This happens even when the robot is disabled. Moving this to a command may help prevent lockups.

azvanderpas, 2022-03-01:
Never ran into a lockup testing on a roboRIO and color sensor after many redeploys at desk setup. Possible reasons:

- Setup is deploying via USB connection, its possible lockups only affect the radio/ethernet interfaces
- CDS is the only subsystem running, and no motor controllers are connected. Less possibility for electrical interference or process scheduling problems.
- Maybe deploying with the same code causes a different method of deploying (i.e. restart rather than kill, copy, start)
- Issue could be with the TCA9548A not present in this setup.

azvanderpas, 2022-03-01:
Something to try to reproduce roboRIO behavior:
Run code exercising the I2C bus and then short SDA to GND. Might not work first try since need to be in the middle of transaction for it to work.

azvanderpas, 2022-03-01:
Also noticing that Rev Robotics' ColorSensorV3 code does exactly zero error checking on their I2C transactions...

AndrewIverson, 2022-03-03:
Alright this might be over our heads here(or at least mine) its possible that its an issue with the FGPA, the Xilinx Zynq-7000 has some known issues that we might be running into

https://www.xilinx.com/support/documentation/errata/en247.pdf if you ctrl+f for deadlock we get a few results that are intresting.

"Strongly Ordered Write Followed By LDREX Might Deadlock Processor" and "A Data Cache Maintenance Operation Which Aborts, Followed By An ISB, Without Any DSB
In-between, Might Lead To Deadlock" both are marked as won't fix. 😆  Both seem to be fixed if you call a DSB inbetween the operations. 

Looking at the [documentation for the 7000](https://www.xilinx.com/support/documentation/user_guides/ug585-Zynq-7000-TRM.pdf) 
```7. Synchronization barriers:
dsb (Allows MMU to start)
isb (Flushes pre-fetch buffer)
(Refer to Memory Barriers, page 73 for more details on memory barriers.```

Now is where I start to get lost, I'm not sure how the FGPA (zynq-7000) is setup to read the I2C port, I don't know what calls it makes. If we find where/how its doing that we should ensure that its calling a DSB before attempting to read it...

azvanderpas, 2022-03-03:
HAL I2C code is here: https://github.com/wpilibsuite/allwpilib/blob/main/hal/src/main/native/athena/I2C.cpp
Its C++ that gets imported to Java elsewhere. The ioctl function is the kernel call to access the I2C. 

I'm doubting this specific lockup issue is the problem, it doesn't appear to be related to I2C and occurs in very specific cases: https://support.xilinx.com/s/article/47549?language=en_US

The I2C issues in that errata seems kind of interesting though.

azvanderpas, 2022-03-03:
Tested MXP I2C port vs Onboard I2C resiliency against shorting SDA to GND. 

The Onboard port starts issuing CommandScheduler loop overrun errors and the bus never recovers after the short is removed. 

The MXP port however behaves much better, no loop overrun errors and the bus fully recovers after the short is removed. Based on this, I recommend trying the MXP I2C port as plan A, and moving to the Raspberry Pi implementation if we still see lockouts as plan B.

azvanderpas, 2022-03-03:
Some more data. I took timestamps before and after a getProximity calls and graphed the deltas. Onboard I2C takes a couple hundred microseconds normally, but once SDA is shorted it jumps up to **100 ms** and stays there even after the short is removed.

[image]

MXP takes a similar couple hundred microseconds normally, but when SDA is shorted it actually decreases (presumably the lockup is detected somehow) and then recovers when the short is removed.

[image]

Probably not super relevant, but the I2C protocol seems to be implemented differently on the ports based on their scope captures. 

Onboard:
[image]

MXP:
[image]

closed 2022-03-29 by azvanderpas