AusTINCANsProgrammingTeam/2022RobotCode #133: CDS Changes for #125

AndrewIverson, 2022-03-03:
#125  its possible always checking the color sensor is causing an issue, move the code out of periodic and into teleop periodic and autonomous periodic.

sarango2158, 2022-03-04:
In the CDS branch @Ariya3 and @zainmfjavaid moved getSensorStatus out of periodic and into the execute() method of their CDSAutoAdvanceCommand and CDSBallManagementCommand (which is the default command of the subsystem). This may be an improvement, but worth checking how often this is called.

caseyjbrotherton, 2022-03-05:
In testing, they were using a println, and this seemed to be still every 20ms.  

Should we try once every loop?  Reduce the frequency down to a max of every 40ms?

sarango2158, 2022-03-15:
tied to runInterval in CDSBallManagement

closed 2022-03-15 by sarango2158