AusTINCANsProgrammingTeam/2022RobotCode #138: Event logging to Shuffleboard

caseyjbrotherton, 2022-03-05:
Shuffleboard can log "events"  This is a string with a time and severity.

We can use this to mark parts of shuffleboard logging that we want to examine further.

caseyjbrotherton, 2022-03-05:
CDS came up with these ideas:

Competition Debugging for CDS -
  How many time reverse intake runs
  Next open sensor
  Ball count
  If auto eject is running
  If sensors are on or off
  If auto eject/advance breaks

Some of them are events, others maybe better as numbers.

caseyjbrotherton, 2022-03-05:
Another useful event combination is when we start running the flywheel and when we launch the ball in the Shooter subsystem

Asher123456789, 2022-03-05:
How many time reverse intake runs- Done with ShuffleBoard with an event
Next open sensor- Done with ShuffleBoard using an event
Ball count- Done with SmartDashboard using an int, already in Main
If auto eject is running- Done with Shuffleboard using an event
If sensors are on or off- Done with Shuffleboard using an event
If auto eject/advance breaks- Done with Shuffleboard using an event
Start running the flywheel- Done with Shuffleboard using an event
Launch the ball-

closed 2022-03-05 by Asher123456789