AusTINCANsProgrammingTeam/2022RobotCode #145: CDS/Color sensor missing causes the robot code to crash.

AndrewIverson, 2022-03-06:
If a color sensor is missing/unplugged the code currently crashes. The cds/color sensor code needs to be fault tolerant, we can't have the whole robot code crashing during a match because a sensor came unplugged.

sarango2158, 2022-03-06:
Since the default command for CDS (BallManagement) also heavily relies on the color sensors, it would be nice if there was a clear indicator if the sensors were down to the operator (and possibly a fallback like running CDSForward for a few seconds).

AndrewIverson, 2022-03-07:
So I think the best way to handle it is similar to how we have driveTestin1 branch setup. If the color sensors are missing or un detected or whatever we go into override mode. Override mode ties the intake and CDS together at the one button. The one button runs the intake forward, cds forward and stopper wheel in reverse.

zainmfjavaid, 2022-03-09:
Hopefully fixed this, will have to test on Wednesday.

closed 2022-03-09 by zainmfjavaid