AusTINCANsProgrammingTeam/2022RobotCode #146: Add command in auto routines to deploy the intake

sarango2158, 2022-03-06:
In order to stay within our frame perimeter our robot will start with the intake velcroed up. We need to briefly run the intake forward to deploy it. Make sure we have this in our auto routines

AndrewIverson, 2022-03-07:
The combination to unwind is run intake forward and run CDS Vertical wheels(singulator) forward to unwind the velcro.

tonyn10, 2022-03-09:
I'll need to talk with the Intake/CDS team. My questions are:

Is there a certain amount of time needed to run each command to have it fully deployed? If so, do the commands incorporate that in some way?

AndrewIverson, 2022-03-09:
It needs to run for about .5 seconds I'd guess.

zainmfjavaid, 2022-03-10:
Pushed command onto intake branch. We have intake being run for .5 seconds right now, but it should be easy to change that if need be.

closed 2022-03-12 by tonyn10