AusTINCANsProgrammingTeam/2022RobotCode #160: Feature: When CDS has two cargo in it start the fly wheel accelerating automatically.

AndrewIverson, 2022-03-10:
If we have two cargo we are probably moving to shoot the cargo. 
We can spool the flywheel up to shooting speed.

It is still desired to have the shoot while held button. The scoring sequence will be faster because there will basically be no waiting once the driver presses the button.

This should probably be a new command, calls setReference on the flywheel pid controller to the shooting speed when the command ends do not call setReference with 0 speed, leave the reference on shooting speed. Should probably be a default command for the shooter.

AndrewIverson, 2022-03-11:
This might not be needed if we tune the fly wheel well enough. According to math and physics the flywheel should take less than a second to hit the needed speed.

Majorferguson21, 2022-03-11:
I will be giving characterization and calculated FFs another shot today, hopefully that helps with tuning. If we have to drop our CDS automatic features, we could add spooling to the operator's set of controls?

AndrewIverson, 2022-03-12:
#162  will make this irrelevant. Shoots in less than a second

closed 2022-03-12 by AndrewIverson