AusTINCANsProgrammingTeam/2022RobotCode #185: CDS to do

Ariya3, 2022-03-15:
- add stopper wheel to combined intake and CDS
- check sensors for good data, if they return 0 run combined cds/intake
    - make sensor statuses private variables

caseyjbrotherton, 2022-03-15:
Notes:

- We want to use the same sensor read for checking sensor status as checking beam break. ( Use instance variables instead of method variables. )
- We want to track the failures.
- When the failures are above a threshold, set isFinished to true
- in end, stop eject.  Stop CDS belt moving forward.  Warn Shuffleboard.
- Should check if instead of end'ing, we can schedule a new command that is mapped to a button.
- ( Or should we always have the command mapped to the button at all times, and have the cargo wheel moving in reverse. )

azvanderpas, 2022-03-29:
In testing, we've only seen brief periods of zero-ed out data, and zeros haven't (and shouldn't be able to) cause false positives. I think we can get away without adding the sensor failure checking. False negatives are acceptable.

Closing this issue since everything listed has either been addressed (stopper wheel), or is no longer relevant.

closed 2022-03-29 by azvanderpas