AusTINCANsProgrammingTeam/2022RobotCode #224: Shooter Feedforward should use setVoltage instead of setReference(velocity)

azvanderpas, 2022-04-04:
Feedforward control: https://docs.wpilib.org/en/stable/docs/software/advanced-controls/controllers/feedforward.html

Feedforward + PID control: https://docs.wpilib.org/en/stable/docs/software/advanced-controls/controllers/combining-feedforward-feedback.html

Majorferguson21, 2022-04-04:
So we should switch to WPIlib PID to use both in setVoltage? 
I thought sparkMax's setReference included the arbFF param as a way to get around using WPIlib PID if we wanted to use FF control, but I could be wrong

azvanderpas, 2022-04-04:
Maybe you are right. Its odd that I don't see any examples of that being used anywhere though.

Either way we need to fix the shooter tuning since it appears we aren't staying stable, or at least we aren't checking that we are. Possibly the setReference doesn't do voltage compensation correctly since it's affected by our belt speed?

Majorferguson21, 2022-04-04:
It is really odd, I was looking for a long time for any sort of documentation or examples, and there was nothing.
I'll work with keeping PID stable, though

caseyjbrotherton, 2022-04-04:
Talks from whiteboard:
  - Need to characterize when we have 2 wheels from manufacturing. ( RPM may need to be higher because of less spin, and less magic forces. )
  -  Need to check and see that RPM holds after sending cargo.  Right now, we use an exponential smoothing, but once we hit that point, we send all balls without rechecking RPM. 
  - Investigate Auton and belt issue.  Worked around it by just sending the ball even if the RPM doesn't hit.  This seemed to work during competition, however, it is a mystery and we should tune belt speed, or something to make sure that auton can launch the ball.